landonr/homeThing #1: Source code is missing

mshulman, 2022-09-04:
I don’t see the c++ code in the repo, only header files. 

Where should I look? 

Also, do you have build files checked in?

landonr, 2022-09-05:
its all in the header files. ill make a pr to move it to cpp files

landonr, 2022-09-09:
it wouldn't compile :I. my other esphome projects all include .h files only so it might be how its supposed to be

mshulman, 2022-09-09:
What hardware should I buy to try this?

-ms

landonr, 2022-09-10:
https://github.com/landonr/esphome-remote/blob/main/GettingStarted.md#what-you-need

An M5Stack Fire is the easiest because it already has a battery and a case. If you end up going with a Lilygo T4 or TDisplay you can add a rotary scrollwheel using the gpio pins

Pretty much any esp32 + display combo with 2+ buttons that works with ESPHome will work for this project

LuxStatera, 2022-09-27:
Will a D1 mini work?

landonr, 2022-09-27:
I have one but I don't have an extra display. It should, which display are you using?

LuxStatera, 2022-09-27:
I have ordered any parts yet since I wanted to see if I would also need to order an ESP32 or if a D1 mini would work since I already have a few extras

landonr, 2022-09-28:
D1 should work with any display supported by esphome

LuxStatera, 2022-09-28:
Sounds good I'll order some parts. Any plans for a 3D printed case?

landonr, 2022-09-28:
yes it's in progress. hopefully just a few weeks

LuxStatera, 2022-09-28:
You rock!

landonr, 2023-04-02:
fixed in #41

closed 2023-04-02 by landonr