landonr/homeThing #3: Watchdog triggering on M5Stack Fire

ralfkah, 2022-09-05:
Hey,

I love your project and tried to install it on my M5Stack Fire.
I already tried it 3-4 times, but the software keeps rebooting with a watchdog triggering.
Logs attached. I don't know how to proceed, can you point me in the right direction?
Not sure how to continue with debugging. Already removed most of my HomeAssistant entities (scenes, scripts, lights, ...).

I flashed it via ESPHome Web, not sure if this makes any difference.

Regards
Ralf

> ets Jul 29 2019 12:21:46
> 
> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:10124
> load:0x40080400,len:5828
> entry 0x400806a8
> [I][logger:243]: Log initialized
> [C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
> [D][esp32.preferences:113]: Saving 1 preferences to flash...
> [D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
> [D][PlayerTV:336]: TV subbed media_player.bravia_kdl_50w829b
> [D][Player:080]: Player subbed media_player.bravia_kdl_50w829b
> [D][speakerGroup:417]: New speaker media_player.esszimmer
> [D][speaker:171]: Sonos Speaker subbed media_player.esszimmer
> [D][Player:080]: Player subbed media_player.esszimmer
> [D][speakerGroup:417]: New speaker media_player.spotify_xyz
> [D][speaker:171]: Sonos Speaker subbed media_player.spotify_xyz
> [D][Player:080]: Player subbed media_player.spotify_xyz
> [W][app:022]: Component <unknown> already registered! (0x3ffba53c)
> [W][app:022]: Component <unknown> already registered! (0x3ffba4e4)
> [W][app:022]: Component <unknown> already registered! (0x3ffba580)
> [W][app:022]: Component <unknown> already registered! (0x3ffba510)
> [I][app:029]: Running through setup()...
> [C][spi:023]: Setting up SPI bus...
> [D][binary_sensor:034]: 'M5_BtnA': Sending initial state OFF
> [D][binary_sensor:034]: 'M5_BtnB': Sending initial state OFF
> [D][binary_sensor:034]: 'M5_BtnC': Sending initial state OFF
> [C][switch.gpio:011]: Setting up GPIO Switch 'Backlight'...
> [D][switch:013]: 'Backlight' Turning ON.
> [D][switch:037]: 'Backlight': Sending state ON
> [D][switch:013]: 'Backlight' Turning ON.
> [C][fastled:012]: Setting up FastLED light...
> [C][light:036]: Setting up light 'Remote Light'...
> [D][light:035]: 'Remote Light' Setting:
> [D][light:040]:   Color mode: RGB
> [C][adc:036]: Setting up ADC 'VBatt'...
> [D][binary_sensor:034]: 'Node Status': Sending initial state OFF
> [C][wifi:037]: Setting up WiFi...
> [I][wifi:248]: WiFi Connecting to 'WiFiName'...
> [D][sensor:127]: 'VBatt': Sending state 1.69641 V with 2 decimals of accuracy
> [D][sensor:127]: 'vcc_adc_avg': Sending state 1.69641  with 1 decimals of accuracy
> [D][sensor:127]: 'batteryVoltage': Sending state 3.00265 V with 1 decimals of accuracy
> [D][sensor:127]: 'Battery Percent': Sending state 0.37844 % with 1 decimals of accuracy
> [D][sensor:127]: 'Battery Percent': Sending state 0.37844 % with 1 decimals of accuracy
> [D][light:035]: 'Remote Light' Setting:
> [D][light:046]:   State: ON
> [D][light:108]:   Effect: 'Rainbow Effect'
> [D][drawing menu:383]: turning on
> [D][switch:013]: 'Backlight' Turning ON.
> [D][sensor:127]: 'VBatt': Sending state 1.70616 V with 2 decimals of accuracy
> [D][idle:412]: turning off display
> [D][switch:017]: 'Backlight' Turning OFF.
> [D][switch:037]: 'Backlight': Sending state OFF
> [D][sensor:127]: 'VBatt': Sending state 1.73282 V with 2 decimals of accuracy
> [D][light:035]: 'Remote Light' Setting:
> [D][light:046]:   State: OFF
> [D][light:108]:   Effect: 'None'
> [D][drawing menu:387]: turning off 20
**> E (26849) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (26849) task_wdt:  - loopTask (CPU 1)
> E (26849) task_wdt: Tasks currently running:
> E (26849) task_wdt: CPU 0: IDLE0
> E (26849) task_wdt: CPU 1: loopTask
> E (26849) task_wdt: Aborting.
> abort() was called at PC 0x40168de8 on core 0
> 
> ELF file SHA256: 0000000000000000
> 
> Backtrace: 0x4008e274:0x3ffbf890 0x4008e4ed:0x3ffbf8b0 0x40168de8:0x3ffbf8d0 0x4008a101:0x3ffbf8f0 0x401874cb:0x3ffbc230 0x4016aecb:0x3ffbc250 0x400916fd:0x3ffbc270 0x4008ff52:0x3ffbc290
> 
> Rebooting...
> ets Jul 29 2019 12:21:46**
> 
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:10124
> load:0x40080400,len:5828
> entry 0x400806a8
> [I][logger:243]: Log initialized
> [C][ota:461]: There have been 1 suspected unsuccessful boot attempts.
> [D][esp32.preferences:113]: Saving 1 preferences to flash...
> [D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
> [D][PlayerTV:336]: TV subbed media_player.bravia_kdl_50w829b
> [D][Player:080]: Player subbed media_player.bravia_kdl_50w829b
> [D][speakerGroup:417]: New speaker media_player.esszimmer
> [D][speaker:171]: Sonos Speaker subbed media_player.esszimmer
> [D][Player:080]: Player subbed media_player.esszimmer
> [D][speakerGroup:417]: New speaker media_player.spotify_xyz
> [D][speaker:171]: Sonos Speaker subbed media_player.spotify_xyz
> [D][Player:080]: Player subbed media_player.spotify_xyz
> [W][app:022]: Component <unknown> already registered! (0x3ffba53c)
> [W][app:022]: Component <unknown> already registered! (0x3ffba4e4)
> [W][app:022]: Component <unknown> already registered! (0x3ffba580)
> [W][app:022]: Component <unknown> already registered! (0x3ffba510)
> [I][app:029]: Running through setup()...
> [C][spi:023]: Setting up SPI bus...
> [D][binary_sensor:034]: 'M5_BtnA': Sending initial state OFF
> [D][binary_sensor:034]: 'M5_BtnB': Sending initial state OFF
> [D][binary_sensor:034]: 'M5_BtnC': Sending initial state OFF
> [C][switch.gpio:011]: Setting up GPIO Switch 'Backlight'...
> [D][switch:013]: 'Backlight' Turning ON.
> [D][switch:037]: 'Backlight': Sending state ON
> [D][switch:013]: 'Backlight' Turning ON.
> [C][fastled:012]: Setting up FastLED light...
> [C][light:036]: Setting up light 'Remote Light'...
> [D][light:035]: 'Remote Light' Setting:
> [D][light:040]:   Color mode: RGB
> [C][adc:036]: Setting up ADC 'VBatt'...
> [D][binary_sensor:034]: 'Node Status': Sending initial state OFF
> [C][wifi:037]: Setting up WiFi...
> [I][wifi:248]: WiFi Connecting to 'WiFiName'...
> [D][sensor:127]: 'VBatt': Sending state 1.69884 V with 2 decimals of accuracy
> [D][sensor:127]: 'vcc_adc_avg': Sending state 1.69884  with 1 decimals of accuracy
> [D][sensor:127]: 'batteryVoltage': Sending state 3.00694 V with 1 decimals of accuracy
> [D][sensor:127]: 'Battery Percent': Sending state 0.99179 % with 1 decimals of accuracy
> [D][light:035]: 'Remote Light' Setting:
> [D][light:046]:   State: ON
> [D][light:108]:   Effect: 'Rainbow Effect'
> [D][drawing menu:383]: turning on
> [D][switch:013]: 'Backlight' Turning ON.
> [D][sensor:127]: 'VBatt': Sending state 1.70025 V with 2 decimals of accuracy

landonr, 2022-09-05:
can you try updating the esphomeRemote.h drawMenu() function with this 
`
void drawMenu() {
  if(idleTime > 16) {
    menuDrawing = false;
    return;
  }
  if (speakerGroup -> playerSearchFinished == false) {
    speakerGroup -> findActivePlayer();
    menuDrawing = false;
    // drawBootSequence();
    return;
  }
...
`
the dumb boot sequence might be taking too long to draw

ralfkah, 2022-09-05:
I changed the header file according to your suggestion. Now the device keeps on, seems that your idea was right regarding the boot sequence. But now the screen stays off (black), although I can see button events in the log file, example below:

> [C][mdns:094]: mDNS:
> [C][mdns:095]:   Hostname: fireremote
> [C][ota:085]: Over-The-Air Updates:
> [C][ota:086]:   Address: fireremote.local:3232
> [C][ota:089]:   Using Password.
> [W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 3 restarts
> [C][api:138]: API Server:
> [C][api:139]:   Address: fireremote.local:6053
> [C][api:143]:   Using noise encryption: NO
> [C][deep_sleep:049]: Setting up Deep Sleep...
> [C][deep_sleep:059]:   Wakeup Pin: GPIO39
> [D][sensor:127]: 'VBatt': Sending state 1.69838 V with 2 decimals of accuracy
> [D][sensor:127]: 'VBatt': Sending state 1.67517 V with 2 decimals of accuracy
> [D][idle:412]: turning off display
> [D][switch:017]: 'Backlight' Turning OFF.
> [D][switch:037]: 'Backlight': Sending state OFF
> [D][light:035]: 'Remote Light' Setting:
> [D][light:046]:   State: OFF
> [D][light:108]:   Effect: 'None'
> [D][drawing menu:387]: turning off 20
> [D][binary_sensor:036]: 'M5_BtnB': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnB': Sending state OFF
> [D][switch:013]: 'Backlight' Turning ON.
> [D][switch:037]: 'Backlight': Sending state ON
> [D][light:035]: 'Remote Light' Setting:
> [D][light:046]:   State: ON
> [D][light:108]:   Effect: 'Rainbow Effect'
> [D][binary_sensor:036]: 'M5_BtnA': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnA': Sending state OFF
> [D][sensor:127]: 'VBatt': Sending state 1.70919 V with 2 decimals of accuracy
> [D][binary_sensor:036]: 'M5_BtnA': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnA': Sending state OFF
> [D][binary_sensor:036]: 'M5_BtnC': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnC': Sending state OFF
> [D][binary_sensor:036]: 'M5_BtnB': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnB': Sending state OFF
> [D][binary_sensor:036]: 'M5_BtnB': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnB': Sending state OFF
> [D][speaker:127]: media_player.bravia_kdl_50w829b play pause
> [D][binary_sensor:036]: 'M5_BtnC': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnC': Sending state OFF
> [D][sensor:127]: 'VBatt': Sending state 1.70889 V with 2 decimals of accuracy
> [D][sensor:127]: 'batteryVoltage': Sending state 2.99041 V with 1 decimals of accuracy
> [D][sensor:127]: 'Battery Percent': Sending state -1.37043 % with 1 decimals of accuracy
> [D][sensor:127]: 'Battery Percent': Sending state -1.37043 % with 1 decimals of accuracy
> [D][esp32.preferences:113]: Saving 1 preferences to flash...
> [D][esp32.preferences:142]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
>

landonr, 2022-09-05:
have you added your device to home assistant in the integrations menu? you might have to set the autoClearState to 1 also
```
void drawMenu() {
  if(idleTime > 16) {
    menuDrawing = false;
    return;
  }
  if (speakerGroup -> playerSearchFinished == false) {
    speakerGroup -> findActivePlayer();
    menuDrawing = false;
    autoClearState = 1;
    // drawBootSequence();
    return;
  }
...
```

ralfkah, 2022-09-06:
Yes, I added it to my local HA instance via the esphome-integration. A couple of switch entities (backlight/remote backlight) appeared, also a battery level sensor and a node status. I also tried deleting and re-adding the fireremote device in HA. 

I added the changed code, but still the same result. According to the logs, the software is running and the device is connected to wifi. Display stays dark all the time. As soon as i press a button, the rgb light on the m5stack fire turns on and off again after a while.

I can also see that I can interact with a media_player instance by pressing buttons:
> [D][binary_sensor:036]: 'M5_BtnB': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnB': Sending state OFF
> [D][speaker:127]: media_player.bravia_kdl_50w829b play pause
> [D][binary_sensor:036]: 'M5_BtnB': Sending state ON
> [D][binary_sensor:036]: 'M5_BtnB': Sending state OFF
> [D][speaker:127]: media_player.bravia_kdl_50w829b play pause

When I'm home later today, I will try to debug to find out, what the issue is with my m5stack fire :-)

landonr, 2022-09-06:
we should see some log like **media_player.55_tcl_roku_tv state changed to playing** when the remote finds the tv. this needs to happen before it gets past the boot screen

heres what I see on boot
```
[C][captive_portal:088]: Captive Portal:
[C][mdns:094]: mDNS:
[C][mdns:095]:   Hostname: fireremote
[C][ota:085]: Over-The-Air Updates:
[C][ota:086]:   Address: fireremote.local:3232
[C][api:138]: API Server:
[C][api:139]:   Address: fireremote.local:6053
[C][api:143]:   Using noise encryption: NO
[C][deep_sleep:049]: Setting up Deep Sleep...
[C][deep_sleep:059]:   Wakeup Pin: GPIO39
[D][sensor:127]: 'VBatt': Sending state 1.71500 V with 2 decimals of accuracy
[D][api:102]: Accepted ::FFFF:AC10:143
[D][binary_sensor:036]: 'Node Status': Sending state ON
[D][api.connection:861]: Home Assistant 2022.8.7 (::FFFF:AC10:143): Connected successfully
[D][time:042]: Synchronized time: 2022-09-06 00:18:57
[D][Player:086]: media_player.55_tcl_roku_tv state changed to playing
[D][WARNING:016]: menu already drawing
[D][Player:112]: media_player.55_tcl_roku_tv name changed to 55" TCL Roku TV
[D][light:035]: 'Remote Light' Setting:
[D][light:046]:   State: OFF
[D][light:108]:   Effect: 'None'
[D][light:035]: 'Remote Light' Setting:
[D][light:046]:   State: ON
[D][light:050]:   Brightness: 50%
[D][light:058]:   Red: 50%, Green: 25%, Blue: 100%
[D][PlayerTV:343]: media_player.55_tcl_roku_tv Player source changed to YouTube
```

ralfkah, 2022-09-07:
Hell yeah, finally it's working. It seems that the tv I used in the TVSetup(...) call is not working reliably. I now changed it to another device and voila - the m5 is booting and showing me the menu! 
Great!

I also reverted all changes you suggested in your first response. Even with the BootSequence-Call active again, the m5 is booting and working correctly.

Seems that it was my fault after all :-) Thanks for your help resolving it. Really appreciate it.

Ralf