landonr/homeThing #7: More settings configurable

abroooo, 2022-10-10:
First of all kudos for this great project!
I've been tinkering with it for a couple of days (tdisplay with rotary wheel) and really am enjoying it especially the rotary wheel 😄 

For my use I made the display and sleep timeouts configurable to be able to easily play with them. In my opinion having them hardcoded is not ideal since more users might want to change these values. What do you think?

Do you welcome PRs? I have limited time but would like to contribute to this project.

landonr, 2022-10-10:
glad you're liking the project, please share some pics!

PRs are welcome, feel free to add the settings to the globals and substitutions and make a pr. In the future it'd be nice to have some of these values adjustable in a menu on the remote

abroooo, 2022-10-10:
This is my prototype. Haven't had time to solder yet ...
[image]

landonr, 2022-10-10:
that's great! we're working on a case that just uses a cable which should be ready in the next few weeks

abroooo, 2022-10-10:
That sounds cool, can't wait to see it!

btw: your todo list mentions dimming of lights. Have you started on that yet? I just started to look into it. Just have to figure out how to get the current brightness and how exactly you structured the menus.

landonr, 2022-10-10:
I haven't started that, but if you want to take that on be my guest!

- we have `esphomeRemoteLight.h` which contains the state for the light. you can include the brightness right under `state_changed` and call it from setup
- you can add a `lightsDetailMenu` in `MenuStates` in `MenuGlobals.h` then fill out all the enums
- instead of selecting the light in `selectMenu()`, set the `activeMenuState` to `lightsDetailMenu`
- draw the menu you want in `drawMenu()`
- the rest of the menu code is in enums so it should give you compiler warnings until it's all set up
let me know how it goes

abroooo, 2022-10-11:
Thx for intro I'll look into it.

Menu logic wise I was thinking of _PressSelect_ to keep as is (on/off) and _PressRight_ to enter light details menu.

landonr, 2022-10-11:
that kind of breaks the ipod menu layout but I have no problem with it. maybe we could do hold select to toggle light and tap to enter menu?

landonr, 2022-10-11:
I want to keep select to always enter menus if possible (it shows a > arrow on the right of the title if theres a menu to enter). We should consider some lights can't do brightness / temperature (light switches) so some lights will only toggle.
So if we did select to enter lightDetail or toggle a light switch. Inside lightDetail we could have a slider for brightness which we could select to toggle quickly. That way it's 2 quick taps on a light with detail or just 1 on a switch

abroooo, 2022-10-11:
Alright, sounds good. 

So then we want:
  * lightMenu:
    * tap light => enter lightDetail and show slider(s)
    * hold light => toggle light (<- do we need this?)
  * lightDetailMenu: (draw slider(s) on the screen)
    * tap light => toggle light
    * rotate => increase/decrease brightness

landonr, 2022-10-11:
hold light => toggle light. i was thinking this might be slower than tapping into the menu

#### lightDetailMenu: (draw slider(s) on the screen)
- how would it work if we had temperature and brightness? 
- the lightDetail menu could be
    - light name (tap to toggle)
    - brightness (tap and scroll)
    - temperature (tap and scroll)
    - rgb color? (tap and scroll)

landonr, 2022-10-11:
by the way are you using Sonos? I'm curious how this works with other media player sources

abroooo, 2022-10-11:
> hold light => toggle light. i was thinking this might be slower than tapping into the menu

A bit slower probably 

> 
> #### lightDetailMenu: (draw slider(s) on the screen)
> - how would it work if we had temperature and brightness? 
> - the lightDetail menu could be
>     - light name (tap to toggle)
>     - brightness (tap and scroll)
>     - temperature (tap and scroll)

That's makes sense for me

>     - rgb color? (tap and scroll)

Hmm I can't imagine that this would be nice to control. We could have three sliders for the colors but for me personally this would be overkill 🙈

abroooo, 2022-10-11:
> by the way are you using Sonos? I'm curious how this works with other media player sources

Yes I'm using Sonos. Sources are empty for me and volume up/down doesn't have any effect. Haven't looked into it yet. 
I only have one Sonos configured _media_player.<sonos_name>_ and also no TV. What provides the sources content in your video?

landonr, 2022-10-11:
> Yes I'm using Sonos. Sources are empty for me and volume up/down doesn't have any effect. Haven't looked into it yet. I only have one Sonos configured _media_player.<sonos_name>_ and also no TV. [...]

ill make a new issue for this

landonr, 2022-10-11:
> Hmm I can't imagine that this would be nice to control. We could have three sliders for the colors but for me personally this would be overkill 🙈

yeah I agree it's overkill but it's kinda cool. the color picker could be a hsv slider for rgb since brightness is separate. Like the top bar here

[image]

abroooo, 2022-10-12:
> yeah I agree it's overkill but it's kinda cool. the color picker could be a hsv slider for rgb since brightness is separate. Like the top bar here
>
> [image]

That would be a total _because we can feature_ and I like it 😆 !

landonr, 2022-10-12:
I think it looks good! It's pretty similar to the clock menu on ipod classic where there's double height rows. https://youtu.be/-USWx1WErMY?t=33

- We could put the light name in the header instead of Light Details
- We could make the brightness, and temperature double height and modify the menu code to support it
- Then you can just select either brightness, or temp and scroll to adjust, then select or menu to exit 
- This double row height menu looks like it should be reusable if we implement sliders for curtains or hvac!

abroooo, 2022-10-12:
Thx! I really like your ideas. I think you meant something like this?::
[image]
How would we turn the light off/on? We need select to select one of the sliders. Press and hold doesn't seem very intuitive here.

There's a lot more to this feature then I initially thought 😆 The light detail menu is another layer added and it consists of a different type (sliders not strings) which also provide a different functionality. And we also need to remember the light selected for use in the light detail menu. Have you already thought about how you want the architecture to develop? 

Maybe we could have a MenuItemBase class and derive MenuTitle and MenuSlider from it. That way we could use `drawMenu(std::vector<MenuTitle>)` for the new menu as well. Haven't thought this through yet though.

I'll think about it 😄

landonr, 2022-10-12:
- for remembering which light selected for detail we could do something like grouping where i just use this `speakerGroup -> newSpeakerGroupParent` to figure out which speaker you want set up. This doesn't seem very extendible because we'd need to set up a variable for each menu. 
- maybe we keep a vector of the MenuTitles (or base menus) that we've entered as a `NavigationStack`. Then we can see what menu/entity we're in by looking at the top of the array, and it would allow us to pop the array to go back a page
- `MenuSlider` with the base is a good idea. It would need min, max, value, entityId, friendlyName, type (brightness, temp, rgb if we want to draw different slider types on screen)

abroooo, 2022-10-13:
> * for remembering which light selected for detail we could do something like grouping where i just use this `speakerGroup -> newSpeakerGroupParent` to figure out which speaker you want set up. This [...]

Yeah, having a stack sounds good. 

> * `MenuSlider` with the base is a good idea. It would need min, max, value, entityId, friendlyName, type (brightness, temp, rgb if we want to draw different slider types on screen)

I think it'll be a bigger refactor and it probably makes sense to split this into multiple PRs? 

> 
> ps for toggling i added this edit in the last comment im not sure if you saw it.
> 
> > i just realized that moving the name to the header would remove the toggle functionality so maybe we should keep it how you have it

Nope I'm sorry I missed that. So then let's keep the switch.

landonr, 2022-10-13:
> - MenuSlider with the base is a good idea. It would need min, max, value, entityId, friendlyName, type (brightness, temp, rgb if we want to draw different slider types on screen)

>I think it'll be a bigger refactor and it probably makes sense to split this into multiple PRs?
how are we going to adjust the light values in this pr?

abroooo, 2022-10-13:
I'm not sure I get what you mean? Did you mean how are we going to get a functional light details menu without refactoring the menu?

I think something like this:
* we can save the currently selected light in `selectMenu()`
* probably have a function `drawMenuLightDetail()` which draws the new menu
* and then react in `selectMenu()` (set selection as well move the slider)
* when a slider is selected we also have to save that to react accordingly

What do you think? When a slider is selected and moved, where do we draw that?

landonr, 2022-10-13:
oh yeah I misunderstood, that sounds good for now!
> What do you think? When a slider is selected and moved, where do we draw that?

for volume there's a localVolumeLevel that updates the volume level of a speaker. the local volume level is kept for the ui because it takes a while for home assistant to reply with the updated state. we can do the same thing for the slider value in lights by keeping a local brightness value and subscribing to the brightness

abroooo, 2022-10-14:
Alright, I got it working. I implemented it like this:

* `ButtonPressNext` and `ButtonPressPrevious` for scrolling the menu
* `ButtonPressSelect` to select either toggle the light or select one of the sliders
* if slider is selected `ButtonPressNext` and `ButtonPressPrevious` to increase/decrease  color temperature / brightness
* if slider is selected `ButtonPressUp` to deselect

Whats left to do:
* draw animated sliders
* draw selection
* tidy up

There's some hard-coded stuff (couldn't use MenuTitle class for the items). When I send the PR we should discuss if we want to leave this for now and do that when we refactor the menu or do it right now.

abroooo, 2022-10-14:
> oh yeah I misunderstood, that sounds good for now!
>
>
> for volume there's a localVolumeLevel that updates the volume level of a speaker. the local volume level is kept for the ui because it takes a [...]

Good to know. I currently have the brightness/color_temp as a class member and increase/decrease it when changing the brightness/color_temp. It's also updated by the callback from the subscription (pretty much as you suggested in your comment). I'm not sure if this can be a problem in terms of data consistency.

I experienced some problems when updating birghtness/color_temp. After a number of successive changes the light got unresponsive. Not sure if it's the light/zigbee network/something else. Just thought I'd mention it because maybe you already know/experienced this and ideally have an idea 😛 If not I'll add to my list of _stuff to look up/find out_ and it would/should be a known limitation because there's nothing we can do from the code.

landonr, 2022-10-14:
it might stop responding if you spam too many calls to the light.. Maybe we should make it only update the light after selecting on the slider for now instead of on scroll to see if it fixes it?

abroooo, 2022-10-17:
> it might stop responding if you spam too many calls to the light.. Maybe we should make it only update the light after selecting on the slider for now instead of on scroll to see if it fixes it?

I'd like to keep the instantly-changing style because that feels better than selecting a brightness and then setting it. If it leads to problems for some users we could make it optional? Or maybe a customizable rate limit?

abroooo, 2022-10-31:
> that's great! we're working on a case that just uses a cable which should be ready in the next few weeks

I'm curious, how's it going?

landonr, 2022-10-31:
> > that's great! we're working on a case that just uses a cable which should be ready in the next few weeks
> 
> I'm curious, how's it going?

i'll follow up on discord

landonr, 2022-11-09:
@abroooo here's a pic of the case so far 
[image]

the rotary, battery and tdisplay mount to the case so we wont need a circuit board, only a cable to connect the rotary. the idea is to keep the price really cheap for this version. once we finish this off it would be fun to make a custom pcb with a different screen and more buttons

theres more pics on discord, i fixed the invite link https://discord.gg/BX6ZtGKHTy

abroooo, 2022-11-10:
@landonr that looks awesome 👊🏻🚀 and also nice that no extra pcb is needed. That makes it more accessible I think.

Will you add the files for the case to the repo?

landonr, 2022-11-17:
we'll add them as soon as we can

closed 2022-11-17 by landonr