landonr/homeThing #13: Sonos Favorites aren't loaded

landonr, 2022-10-11:
**Describe the bug**
Sonos favorites are empty because they rely on a home assistant template

**Expected behavior**
All the sonos favorites should be listed in the Sources menu

**Additional context**
I use this template to get favorites formatted for the device but it should load the existing sensor.sonos_favorites 
```
      - name: Sonos Favorites List
        state: 'OK'
        attributes:
          items: '{{ state_attr("sensor.sonos_favorites", "items").values() | list | join("~") }}'
```

abroooo, 2022-10-12:
Oh nice, the workaround did the trick for me 👍 
I don't mind the extra sensor but you're right, it's an external dependency.

landonr, 2022-10-12:
If we can get rid of the sensor, we should be able to load playlists from spotify too. Spotcast has this playlists sensors that returns the same as the default sonos sensor
```
playlists: 
- uri: spotify:playlist:37i9dQZF1EIdNK9vIbNBMb
  name: Morning Mix
- uri: spotify:playlist:4772bP5KCO80HXMQsr7aol
  name: S/S 22 🫶🏻
...
```

landonr, 2022-10-14:
https://github.com/landonr/esphome-remote/pull/14

landonr, 2022-10-15:
just need to add docs

abroooo, 2022-10-17:
yeees 😄

abroooo, 2022-10-19:
docs look fine. I'm nevertheless struggling to get it working but that has nothing to do with your docs. I have a `sensor.playlists_sensor` but it does only contain a few of my playlists. I can see these in the Sources menu on my remote so that's working fine 😄 
Good job 👍

closed 2022-10-20 by landonr