landonr/homeThing #16: Hi funny project! thanks a lot!

goxofy, 2022-10-18:
Is it possible to support this hardware?

LILYGO® TTGO T-Display Keyboard For LNURLPoS

https://m.tb.cn/h.UeTxbek?tk=sYnC2zqligT CZ0001

thanks a lot!

landonr, 2022-10-18:
Yes it would be very easy to set up. The code for the rotary is 5 buttons (up,down,left,right,select) and 2 rotary directions (left,right). This keypad has 12 buttons + so you can easily just call the existing functions from the keypad buttons

goxofy, 2022-10-19:
Hi @landonr , i am not a programer, could you please tell me which src file should I change? thanks!

landonr, 2022-10-19:
what do you want each button to do?

landonr, 2022-10-20:
here's the functions you have to call for each button
```
5 - buttonPressSelect();
4- buttonPressLeft();
6 - buttonPressRight();
8 - buttonPressDown();
2 - buttonPressUp();

# - buttonPressNext();
* - buttonPressPrevious();
```
in order to actually set up that keypad this component might be helpful https://github.com/Syralist/esphomekeypad. You can include that and then call the appropriate function based on which number is pressed

goxofy, 2022-10-20:
@landonr ok thanks, i'll try this

landonr, 2022-10-20:
just noticed this component which is better supported should be more straightforward to set up https://github.com/ssieb/custom_components/tree/vbus/components/matrix_keypad

goxofy, 2022-10-21:
OK，thanks!

goxofy, 2022-10-31:
hi @landonr , I have a question about tdisplay with Scrollwheel, what the Scrollwheel's enca pin and encb pin connect to?

landonr, 2022-10-31:
I have it set up on pin 12 and 13. that side has enough IO for all of the buttons and the encoder. here's the esphome setup
```
sensor:
## Rotary
  - platform: rotary_encoder
    filters:
    - throttle: 0.01s
    name: "Rotary Encoder"
    internal: true
    pin_a:
      number: 12
      mode:
        input: true
        pullup: true
    pin_b:
      number: 13
      mode:
        input: true
        pullup: true
    id: rotary
    on_clockwise:
      - lambda: |-
          buttonPressWakeUpDisplay();
          buttonPressNext();
    on_anticlockwise:
      - lambda: |-
          buttonPressWakeUpDisplay();
          buttonPressPrevious();
```

goxofy, 2022-11-01:
hi @landonr 

I made this remote with scrollwheel but when I power on the tdisplay, it only shows a logo, it doesn't work when i press any key.

the logs here, please help me

[image]

landonr, 2022-11-01:
> hi @landonr 
> 
> I made this remote with scrollwheel but when I power on the tdisplay, it only shows a logo, it doesn't work when i press any key.
> 
> the logs here, please help me
> 
> [image]
> 
>  

Have you set up the device in home assistant?

goxofy, 2022-11-01:
yes， I set up it in home assistant

[image]

landonr, 2022-11-02:
have you configured the media players in the `sharedRemoteConfig.yaml` to match your setup? It will show the boot screen until it finds a media player so the UI doesn't have empty states

goxofy, 2022-11-03:
Hi @landonr , thanks for your reply!

I dont have a media player, I just want to control my lights and scene and scripts in home assistant ,so I deleted some code in the `sharedRemoteConfig.yaml`, here is my yaml , is it right?

```
logger:
  level: INFO

api:
  services:
    - service: goto_screen
      variables:
        screen_name: string
      then:
      - lambda: |-
          goToScreenFromString(screen_name);

ota:

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_password}
  fast_connect: true

  ap:
    ssid: "${friendly_name} Fallback"
    password: ${wifi_fallback_password}

captive_portal:

http_request:
  useragent: esphome/device
  timeout: 10s

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 75%
    blue: 10%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: color_accent_primary
    red: ${color_accent_primary_red}
    green: ${color_accent_primary_green}
    blue: ${color_accent_primary_blue}
  - id: my_gray_dark
    red: 40%
    green: 40%
    blue: 40%
  - id: my_gray_dark_2
    red: 8%
    green: 8%
    blue: 8%
  - id: my_gray
    red: 10%
    green: 10%
    blue: 10%
  - id: my_black
    red: 0%
    green: 0%
    blue: 0%
  - id: my_white
    red: 100%
    green: 100%
[... 218 lines cut ...]
     App.register_component(sceneGroup);
     return {sceneGroup};
   id: scene_group_component

 ## Home assistant lights. Set with a friendly name
 - lambda: |-
     std::vector<FriendlyNameEntity> lights { 
       FriendlyNameEntity("light.yeelink_lamp2_ddab_light", "yeelink"), 
     };
     lightGroup->setup(lights);
     App.register_component(lightGroup);
     return {lightGroup};
   id: light_group_component

 ## Home assistant sensors. Set with a friendly name
 - lambda: |-
     std::vector<FriendlyNameEntity> sensors {
       FriendlyNameEntity("sensor.zm1_b0f893254433_temperature", "Ketingwendu"),
     };
     sensorGroup->setup(sensors);
     App.register_component(sensorGroup);
     return {sensorGroup};
   id: sensor_group_component

```

landonr, 2022-11-03:
I'll have to make a fix where it doesn't care about media players! for now you can add `playerSearchFinished = true;` to the top of `void findActivePlayer(bool background = false) {` in esphomeRemotePlayer.h

goxofy, 2022-11-03:
@landonr Thank you very much!!!! 

I will try it and waiting for your patch!

goxofy, 2022-11-03:
Hi @landonr ,I have tried to add `playerSearchFinished = true;` but it also doesn't work,here is log

[image]

and here is the code

[image]

It seems like when I add `playerSearchFinished = true;`, the devices doesn't connect to the HA API because when I click the "Sleep Toggle" it shows 
[image]

landonr, 2022-11-03:
can you share your full config? it seems like its getting disconnected from the api repeatedly. It might be trying to read something that doesn't exist. this thread might help https://github.com/esphome/issues/issues/3108#issuecomment-1216338545 also try increasing your log level to very_verbose and see what it says https://github.com/esphome/issues/issues/3108#issuecomment-1059764294

goxofy, 2022-11-03:
Hi @landonr , 

This is `tdisplayremote.yaml` and I didn't change anything.
```
esphome:
  name: displayremote
  platform: ESP32
  board: featheresp32
  includes:
    - esphomeRemote/MenuGlobals.h
    - esphomeRemote/MenuTitle.h
    - esphomeRemote/FriendlyNameEntity.h
    - esphomeRemote/TextHelpers.h
    - esphomeRemote/esphomeRemotePlayer.h
    - esphomeRemote/esphomeRemoteService.h
    - esphomeRemote/esphomeRemoteLight.h
    - esphomeRemote/esphomeRemoteSensor.h
    - esphomeRemote/DisplayUpdateInterface.h
    - esphomeRemote/esphomeRemote.h
    - esphomeRemote/esphomeRemoteRotary.h

substitutions:
  boot_name: '"TDisplay Remote <3"'
  friendly_name: "TDisplay Remote"
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_fallback_password: !secret wifi_password
  large_font_size: "24"
  medium_font_size: "15"
  small_font_size: "14"
  icon_size: "18"
  icon_size_large: "19"
  header_height: "16"
  margin_size: "4"
  scroll_bar_width: "6"
  bottom_bar_margin: "1"
  now_playing_max_lines: "5"
  draw_now_playing_menu: "false"
  draw_battery_level: "true"
  draw_volume_level: "false"
  draw_shuffle_disabled: "false"
  draw_header_time: "true"
  color_accent_primary_red: 0%
  color_accent_primary_green: 45%
  color_accent_primary_blue: 68%
  display_timeout: "16"
  sleep_after: "7200"
  menu_rollover_on: "false"
  sync_active_player: "true"
  dark_mode: "true"

<<: !include esphomeRemote/sharedRemoteConfig.yaml
  
sensor:
## Rotary
  - platform: rotary_encoder
    filters:
    - throttle: 0.01s
    name: "Rotary Encoder"
    internal: true
    pin_a:
      number: 12
      mode:
        input: true
        pullup: true
    pin_b:
      number: 13
      mode:
        input: true
        pullup: true
[... 510 lines cut ...]
     sceneGroup->setup(scenes);
     App.register_component(sceneGroup);
     return {sceneGroup};
   id: scene_group_component

 ## Home assistant lights. Set with a friendly name
 - lambda: |-
     std::vector<FriendlyNameEntity> lights { 
       FriendlyNameEntity("light.yeelink_lamp2_ddab_light", "yeelink"), 
     };
     lightGroup->setup(lights);
     App.register_component(lightGroup);
     return {lightGroup};
   id: light_group_component

 ## Home assistant sensors. Set with a friendly name
 - lambda: |-
     std::vector<FriendlyNameEntity> sensors {
       FriendlyNameEntity("sensor.zm1_b0f893254433_temperature", "Ketingwendu"),
     };
     sensorGroup->setup(sensors);
     App.register_component(sensorGroup);
     return {sensorGroup};
   id: sensor_group_component
```

goxofy, 2022-11-03:
Does it I must use a battery?

landonr, 2022-11-04:
thanks for the config! I haven't considered using the remote without media player stuff so I'll try to get it working tonight.
no battery isn't required. you can hide the battery level with the draw_battery_level. you need to keep the battery_percent sensor but it can be whatever value.
by the way being able to call the elevator is so cool!

goxofy, 2022-11-04:
Hi @landonr ，I tried to connect a battery and now it works!!

I am waiting for your pr and maybe I could pull a request to support the switches in HA?

Call elevator:

https://user-images.githubusercontent.com/1695416/199864543-cc822276-efb9-49f4-9de3-5a75423934c2.mp4

goxofy, 2022-11-04:
The device is unstable , I must reboot devices for  many times then it will be enter main menu, here is some logs

[image]
[image]

landonr, 2022-11-04:
ill check it out after work sorry I didn't get to this last night. cool elevator video I wish i could figure out how to get my elevator to get called from an api!

landonr, 2022-11-05:
can you rename this issue to something like 'remote wont boot without media players configured'. I made this PR to get started on this https://github.com/landonr/esphome-remote/pull/19

I'll look at the battery issue in a bit

goxofy, 2022-11-05:
ok of course!

landonr, 2022-11-05:
> Hi @landonr ，I tried to connect a battery and now it works!!
> 
> I am waiting for your pr and maybe I could pull a request to support the switches in HA?

absolutely if you want to help out! it should be pretty similar to the basic light component in esphomeRemoteLight.h

I think you can also work around this by just setting up your switch as a basic light in the config

goxofy, 2022-11-06:
Hi @landonr , I tried to support the switches but failed.. I am not a programmer.:)

goxofy, 2022-11-07:
Hi @landonr , I made  a breakout PCB for tdisplay and scroolwheel,like this

[image]
[image]
[image]

landonr, 2022-11-07:
@goxofy that's awesome! I'm going to make a new issue for the breakout PCB so we can close this up

landonr, 2022-11-07:
> Hi @landonr , I tried to support the switches but failed.. I am not a programmer.:)

no worries! I'll make an issue for this and work on it when I have time

closed 2022-11-07 by landonr