landonr/homeThing #20: Cool Project! Could it support Chinese? Thanks

sylvia69999, 2022-11-07:
If I want to use chinese on my remote, what shoud I do?

Thanks

landonr, 2022-11-07:
yes you should be able to replace the characters here with the ones you're using and then build the project. Let me know how it goes https://github.com/landonr/esphome-remote/blob/main/esphomeRemote/sharedRemoteConfig.yaml#L73

sylvia69999, 2022-11-09:
```
  - file: "fonts/font.ttf"
    id: large_font
    size: ${large_font_size}
    glyphs: '儿童房台灯*/\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓëËüÜïÏṣṢ''’'
  - file: "fonts/font.ttf"
    id: medium_font
    size: ${medium_font_size}
    glyphs: '儿童房台灯*/\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓëËüÜïÏṣṢ''’'
  - file: "fonts/font.ttf"
    id: small_font
    size: ${small_font_size}
    glyphs: '儿童房台灯*/\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓëËüÜïÏṣṢ''’'
```

```
FriendlyNameEntity("light.ertongfang", "儿童房台灯"), 
```

sylvia69999, 2022-11-09:
sorry, I don't know how to upload the TTF file.

landonr, 2023-09-12:
this should be easy now that gfonts and online fonts can be downloaded on build

closed 2023-09-12 by landonr