landonr/homeThing #21: Breakout PCB for tdisplay and scroolwheel

landonr, 2022-11-07:
Hi @landonr , I made  a breakout PCB for tdisplay and scroolwheel,like this

[image]
[image]
[image]

_Originally posted by @goxofy in https://github.com/landonr/esphome-remote/issues/16#issuecomment-1304970234_

landonr, 2022-11-07:
@goxofy first off this is awesome work, thank you!

How would we connect the battery to the tdisplay? I think we can add a pin on the back of the pcb for vin/ground and include a battery header

goxofy, 2022-11-14:
I am designing a power mnagement IC for the board in order to charge by the typec on ESP.

landonr, 2022-11-14:
the tdisplay includes a power manager ic. are you sticking with the tdisplay?

goxofy, 2022-11-14:
i am trying to change the builtin ic, now the battery charing is inconvenient

landonr, 2022-11-14:
ah that makes sense. The tdisplay screen is really small. would it make sense to upgrade to a standalone display that's larger like a ILI9341 and use a different esp32 chip?

goxofy, 2022-11-15:
Or can we add another power ic on the board to charge the battery standalone?

landonr, 2022-11-15:
which ic are you looking at? it would be nice to have one with an accurate fuel guage

goxofy, 2022-11-15:
I bought this https://detail.tmall.com/item.htm?id=677456511513&spm=a1z09.2.0.0.4c4f2e8dfx6BFo&_u=f49o7j09dd1 still on the way

landonr, 2022-11-15:
I cant see that without signing up and I don't understand the language. Can you share a screenshot of the specs?

landonr, 2023-03-27:
@goxofy can you share the pcb

closed 2023-10-22 by landonr