landonr/homeThing #29: 3d printed case

landonr, 2022-11-17:
**Is your feature request related to a problem? Please describe.**
the bare components are too pokey

**Describe the solution you'd like**
a 3d printed case where the components are mounted by screws. The case could have pogo pins to charge easily. 

**Describe alternatives you've considered**
we've considered building the case around a PCB but since the components are basic we can skip that and wire everything together using jumper cables 

**Additional context**
requirements
- fits tdisplay
- fits a lipo
- fits ano rotary controller
- doesn't melt while charging
- passthrough buttons for tdisplays internal buttons
- usbc port hole

landonr, 2022-11-17:
here's two different concepts from Jasper who posted in Discord

[image]
[image]

[image]
[image]

abroooo, 2022-11-18:
Great job in designing these case 🎉 
I personally would love to have something small (maybe think of ipod nano or just your protoype with a case around it). Then maybe we don't need pogo pins and just break out the onboard usbc port out?

landonr, 2022-11-18:
this case is designed around a 2200mah lipo so it's about as small as it could be. I don't think the pogo pins are mandatory so we could make a version with or without at this same size

JasperCohen, 2022-11-18:
[image]

This isn't the final case but i think i will be moving forward with some of the aesthetic features, like the front profile and the back shape. I'm still massaging the screen, and the front surface.

JasperCohen, 2022-11-18:
[image]
Also, i know its weird placement but i think this could be a decent spot for a pogo connector. I say this because its dead space right now.

landonr, 2022-11-19:
That looks so good! have you tried printing any prototypes yet? I think for the pogo pin we'd need to make a dock so we can put the pogo pin wherever it works

abroooo, 2022-11-19:
> this case is designed around a 2200mah lipo so it's about as small as it could be. I don't think the pogo pins are mandatory so we could make a version with or without at this same size

ahh I see, also can see it on @JasperCohen cross section drawing. And seeing the pictures with transparent case it doesn't look that big at all 😄  Great work @JasperCohen 🚀

JasperCohen, 2022-11-20:
@landonr this is really just the surfacing so the case isn't ready for printing yet, but at this point I'm going to move ahead with this design with only minor tweaks to the screen trim. 

my goal is to get a printable solid design by today!

landonr, 2022-11-21:
@JasperCohen thanks for adding the reverse engineered parts #32 and for sharing this on discord! 
[image]
this looks super professional take your time!

landonr, 2023-05-01:
case is ordered from our new sponsor, PCBWay! 
[image]

closed 2023-05-01 by landonr