landonr/homeThing #30: Graphs for sensors

abroooo, 2022-11-18:
**Is your feature request related to a problem? Please describe.**
Sensors currently only show numeric values.

**Describe the solution you'd like**
By pressing the sensor menu item it could show a graph over the last x min/hours

**Describe alternatives you've considered**
Instead of drawing it could be a solution to receive a png (from HA) and display that. I wouldn't prefer showing pngs because it would require external configuration and I'm not sure it's possible with our device.

**Additional context**
I investigated a bit and it;s it pretty doable to show a graph. The tricky part is getting the data to show. Natively esphome only supports collecting values over a period of time and show them in a graph. But I would like to see a graph for a sensor even if I just turned on the remote (or return it from deep sleep). Maybe this could be resolved by having ha send an array of values on demand but it would mean external configuration. AFAIK ha doesn't support that natively.

abroooo, 2022-11-19:
Oh the ha rest api, nice idea - havent thought of set to be honest. 
I don't think we need to implement our own graph. `esphome::display::DisplayBuffer::graph()` takes a pointer to an `esphome::graph::Graph` object. We can build this object ourselves.

landonr, 2022-11-22:
are you using a long lived access token? https://developers.home-assistant.io/docs/auth_api/#making-authenticated-requests

abroooo, 2022-11-22:
Yes I am

abroooo, 2022-11-23:
After poking through the esphome code to no avail I found the error - it was the token 🤦‍♂️  I somehow missed a character. After fixing the token it works fine....

landonr, 2022-11-24:
nice! pics

abroooo, 2022-11-25:
yeah I'm pretty much at the beginning so there aren't any pics yet - sorry 😞 

I think I'll approach it like this:

- parse json into array
- build `esphome::graph::HistoryData` object
- build `esphome::graph::GraphTrace` object
- build `esphome::graph::Graph` object
- show `esphome::graph::Graph` obect via `esphome::display::DisplayBuffer:Graph`

landonr, 2023-01-24:
this is really interesting
https://github.com/esphome/esphome/pull/3255#issuecomment-1397300681
this person loads a graph as a png created from https://weewx.com/
I think loading the json from history and drawing the graph on the device is the right move, but this is an interesting approach/hack

abroooo, 2023-01-26:
oh yeah that's nice indeed. I agree, pulling the data from HA and visualizing it on device is the way to go.
btw. it will take me some more time as life keeps me pretty busy atm 😐