landonr/homeThing #35: RGB Lights

landonr, 2022-11-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

landonr, 2022-11-25:
@abroooo what do you think about this for a change to the light detail menu? It shows the light name in the header and the toggle (shows On) previews the light's color
[image]

abroooo, 2022-11-25:
@landonr I'm not sure I understand correctly. IIUC the light in your image is in color mode. If it is in color mode than I think we don't need to show the color temperature slider because it has no effect in this mode. When in color mode we could have something like this maybe?:

[image]
But I like having the title in the header and the toggle just saying on/off.

The color slider would behave like the other sliders. Then we wouldn't need to show a color preview in the toggle. 

How would we switch between color mode and color temp mode? 
We could: 

- have another toggle for switching the modes (I'm not really into that idea)
- use buttonPressDown when color slider selection is active to toggle modes
- something else?

landonr, 2022-11-25:
- yeah I think I see what you're saying. there's no way to even show a color slider preview if the light is in temp mode (and visa versa). 
- I like the idea of showing a preview of the light because it seems fun to code (and I think the base lights menu would look cool if all the toggles had color previews)
- what if the on was an option select (with `off`, `color`, `temp` as the options) similar to tempo in this gem menu [image]

btw is there a way to contact you more directly? could you join the discord or email me (in my profile). I would have just messaged you about this instead of making an issue if i had the option

abroooo, 2022-11-25:
> * yeah I think I see what you're saying. there's no way to even show a color slider preview if the light is in temp mode (and visa versa).

exactly

> * I like the idea of showing a preview of the light because it seems fun to code (and I think the base lights menu would look cool if all the toggles had color previews)

yeah I made a prototype and it was indeed fun. If we do it like in my picture above we'll have two-hundred-something colors which I guess are enough? If not we would need to scroll the color bar or sth.

> * what if the on was an option select (with `off`, `color`, `temp` as the options) similar to tempo in this gem menu [image]

I thought about that as well and I think I wouldn't do that because the light would probably flicker or change color when you just want to switch it off. You know what I mean?

> 
> btw is there a way to contact you more directly? could you join the discord or email me (in my profile). I would have just messaged you about this instead of making an issue if i had the option

I send you a message 😄

landonr, 2022-11-25:
> yeah I made a prototype and it was indeed fun. If we do it like in my picture above we'll have two-hundred-something colors which I guess are enough? If not we would need to scroll the color bar or sth.

we wouldn't have to code all of the colors. the slider position is 0-360 and we can do hsv to rgb conversion. brightness would be the "value" and slider is the hue. saturation is always 100%. https://www.rapidtables.com/convert/color/hsv-to-rgb.html

> I thought about that as well and I think I wouldn't do that because the light would probably flicker or change color when you just want to switch it off. You know what I mean?

we could make it so it wouldnt set the state until you hit center again. so if its on itd be `select` -> `scroll up to "off` -> `select` to confirm

abroooo, 2022-11-29:
> we wouldn't have to code all of the colors. the slider position is 0-360 and we can do hsv to rgb conversion. brightness would be the "value" and slider is the hue. saturation is always 100%. [...]

oh that's nice, I didn't know that! 
Would that be feasible for the user? Because you can to know how hsv works in order to adjust the light to the desired color. In HA you just have a scroll wheel.

The RGB slider would be easy to code as well but not as easy as the hsv stuff.

> we could make it so it wouldnt set the state until you hit center again. so if its on itd be `select` -> `scroll up to "off` -> `select` to confirm

Hmm I think from a users perspective I wouldn't like that because there are extra actions required when I just want to turn off the light. Or did I get it wrong?

landonr, 2022-11-30:
`Hmm I think from a users perspective I wouldn't like that because there are extra actions required when I just want to turn off the light. Or did I get it wrong?`
yeah youre right. What if center always toggled the light, and the option button (under the usb) entered the light detail menu? We could add a different icon to the right of the row to indicate there's an option menu to enter. 
or tapping and holding on one of the rows?

landonr, 2022-12-01:
for the selector this is what i'm thinking. still need to figure out how to toggle the light as easily a regular light
[image]

closed 2023-01-09 by landonr