landonr/homeThing #62: after 'homeassistant_light_group' moved, getting errors while compiling

s3rdl, 2023-06-09:
After the component 'homeassistant_light_group' moved in the last 7 hours, compiling will break.
Last night it was all ok. 
After digging on git, I saw that this component ('homeassistant_light_group') moved to 'homeassistant_common_light'.

You can see the terminal output:

```INFO ESPHome 2023.5.5
INFO Reading configuration homeThing/example-homeThing.yaml...
INFO Updating https://github.com/landonr/homeThing@main
INFO Updating https://github.com/landonr/homeThing@None
INFO Updating https://github.com/landonr/esphome-components@None
Failed config

external_components: [source homeThing/example-homeThing.yaml:172]
  - source: 
      type: git
      url: https://github.com/landonr/homeThing
    refresh: 0s
    components: 
      - homeThing
  - source: 
      type: git
      url: https://github.com/landonr/esphome-components
    refresh: 0s
    components: 
      - homeassistant_component
      - homeassistant_switch_group
      - homeassistant_sensor_group
      
      Could not find __init__.py file for component homeassistant_light_group. Please check the component is defined by this source (search path: homeThing/.esphome/external_components/7e79bc7f/components/homeassistant_light_group/__init__.py).
      - homeassistant_light_group
      - homeassistant_media_player
      - homeassistant_service_group
      - media_player_source
      - media_player_source_sonos
```

Will be there an update in the YAML-files in your 'homeThing' repo?
I was so happy to make further steps, now I am a little bit stuck...

Thank you
Cheers
serdal

sbur83, 2023-06-10:
While this will likely catch us off guard it's a good move. Adding components in YAML will be more user friendly than groups in local folder and fits well with screens.

landonr, 2023-06-14:
I prefer having the folders for sharing stuff between remotes but I get that. I shouldn't be committing my personal yaml here - it should all be example yamls in this repo

closed 2023-06-21 by landonr