landonr/homeThing #70: Lilygo T-Display-S3 - Touch screen doesn't operate menu

sammcj, 2023-07-30:
**Describe the bug**

Howdy, I have a T-Display-S3 - homeThing installs fine and boots, but the menu doesn't respond to the touch screen.

Additionally the two physical buttons do not seem to do any other than wake the screen.

The touch screen itself is working as I can see in the logs it's correctly registering touches.

**To Reproduce**
Steps to reproduce the behaviour:
1. Install homeThing on T-Display-S3 using example config.
2. Upon boot try using the menu.

**Expected behaviour**

To be able to navigate the menu on the touch screen by touching the items and/or using directional swiping gestures.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: homeThing
 - Browser N/A?
 - Version 0.0.6

**Smartphone (please complete the following information):**

Don't see how this is relevant.

**Additional context**

Config file as follows.

```yaml
esphome:
  name: esp-s3
  platformio_options:
    board_build.f_flash: 80000000L
    board_upload.flash_size: 16MB
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S3 Fallback Hotspot"
    password: "zQ9tuPKIfFMu"

time:
  - platform: homeassistant
    id: ha_time

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      inverted: true
    name: "Button 1"
  - platform: gpio
    pin:
      number: GPIO14
      inverted: true
    name: "Button 2"

i2c:
  scl: 17
  sda: 18

font:
  - file: "gfonts://Roboto"
    id: roboto
    size: 30

display:
  - platform: tdisplays3
    id: my_display
    rotation: 90
    update_interval: 10s
    lambda: |-
      id(homeThingMenu)->draw_menu_screen();
      return;

touchscreen:
  - platform: tdisplays3
    interrupt_pin: 16
    address: 0x15
    x_offset: 40
    id: my_touchscreen
    on_touch:
      - lambda: |-
          ESP_LOGI("tdisplays3-touch", "x:%d, y:%d",
            id(my_touchscreen).x,
            id(my_touchscreen).y
          );

external_components:
  - source: github://landonr/lilygo-tdisplays3-esphome
    components: [tdisplays3]
    refresh: 0s
  - source: github://landonr/homeThing
    refresh: 0s
    components: [homeThing] # homething menu
  - source: github://landonr/esphome-components
    refresh: 0s

packages:
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [
        common/device_base.yaml, # defines api, ota, free memory and uptime sensor
        common/ipod/lilygo_tdisplay_ipod_battery.yaml, # used for battery percent
        common/ipod/lilygo_tdisplay_ipod_binary_sensor.yaml, # used for button controls
        common/ipod/lilygo_tdisplay_ipod_rotary.yaml, # used for rotary controls
        common/ipod/lilygo_tdisplay_ipod_sleep.yaml, # required for device to sleep
        common/tdisplay-s3/backlight.yaml, # backlight
        common/fonts.yaml, # default font
        common/icon_fonts.yaml, # material icons
        common/settings.yaml, # settings screen
      ]
    refresh: 0s

substitutions:
  friendly_name: "s3 homething"
  rotary_left_pin: "2"
  rotary_right_pin: "10"
  rotary_up_pin: "3"
  rotary_down_pin: "1"
  rotary_select_pin: "11"
  rotary_pin_a: "13"
  rotary_pin_b: "12"
  screen_left_pin: "GPIO0"
  screen_right_pin: "GPIO14"
  battery_pin: "4"

color: # setup primary accent colour
  - id: my_primary_accent
    red_int: 75
    green_int: 45
    blue_int: 209

light:
  # backlight
  - platform: monochromatic
    output: gpio38
    name: "Backlight"
    restore_mode: RESTORE_DEFAULT_ON
  # home assistant lights
  - platform: homeassistant_component
    id: light_livingroom
    entity_id: light.livingroom
    name: Livingroom Light Group
  - platform: homeassistant_component
    id: light_study
    entity_id: light.study
    name: Study Light Group
  - platform: homeassistant_component
    id: light_bedroom
    entity_id: light.bedroom
    name: All Lights

homeThing:
  id: homeThingMenu
  settings:
    sleep_after: 14400
  sleep_switch: sleep_toggle
  backlight: backlight
  battery:
    battery_percent: battery_percent
    charging: charging
  # media_player_group: media_group_component
  display: my_display
  on_redraw:
    then:
      component.update: my_display
  display_state:
    colors:
      accent_primary: my_primary_accent
    draw_battery_level: true
    font_small: small_font
    font_medium: medium_font
    font_large: large_font
    font_large_heavy: large_heavy_font
    font_material_large: material_font_large
    font_material_small: material_font_small
    font_logo: home_thing_logo
  screens:
    - name: Lights Screen
      entities:
        - id: light_livingroom
          type: light
        - id: light_study
          type: light
        - id: light_bedroom
          type: light

        - type: command
          name: "bedroom lights"
          command:
            - homeassistant.service:
                service: light.toggle
        - type: command
          name: "lights off"
          command:
            - homeassistant.service:
                service: light.turn_off
        - type: command
          name: "lights on"
          command:
            - homeassistant.service:
                service: light.turn_on
    - name: Settings Screen
      show_version: True
      entities:
        - id: backlight
          type: light
        - id: "restart_switch"
          type: switch
        - id: wifi_ssid
          type: text_sensor
        - id: wifi_signal_percent
          type: sensor
        - id: wifi_ip
          type: text_sensor
```

landonr, 2023-07-31:
I haven't actually done any work with touch screens. Are you interested in using this with a touch screen? I figured the device would be too small

You can keep the substitution for `screen_left_pin` and right and remove your duplicate binary sensor

sammcj, 2023-08-10:
Ahhhh ok sorry I didn't realise that!

I was was interested to try it out as I'm looking at options for a home-assistant remote, something to trigger scenes, adjust lights etc... and your project looked like it might make it easy for me to do that. But no worries at all if it was never built with touch in mind I'll have a hack on something myself.

Thanks :)

closed 2023-08-10 by sammcj