landonr/homeThing #78: Hi , I have a proble about rotary

goxofy, 2023-09-19:
**Describe the bug**
I found this project long long ago, I use a Lilygo T-Display TTGO with Adafruit ANO Scrollwheel for this project. It works well.

After an esphome update recently, my remote doesn't work, it seems like the esphome changed the something about the fonts, it report bug while complie, so I tried to use the latest main branch to flash my device.

While I flashed successfully and it works well except the rotary, I rotated it but the screen has nothing changed. So I have reviewd the code and compare the latest code with my old one, and I found the rotary PIN is changed in latest branch, eg:

                                    latest                   longlong ago old
  rotary_pin_a                 21                            12
  rotary_pin_b                 22                           13
  rotary_left_pin              15                            21
  rotary_right_pin           12                            17
  rotary_up_pin                2                            22
  rotary_down_pin           13                           2
  rotary_select_pin         17                           15

 I tried chane the PIN in my configuration with the `substitutions` code, but it still doesn't work.

So what I want to know is whether there are any other changes to this rotary?

Thanks a lot!

landonr, 2023-09-20:
I'm not sure why it's not working. Here's an up to date config with substitutions. https://github.com/landonr/homeThing/blob/main/tdisplay-s3.yaml

Can you looking into this file to see if anything needs changing for your config? You can copy it into your config if you comment out the package include line https://github.com/landonr/homeThing/blob/main/common/ipod/lilygo_tdisplay_ipod_binary_sensor.yaml

landonr, 2023-10-05:
@goxofy did you get anywhere with this?

goxofy, 2023-10-05:
sorry I havent done with this.

closed 2023-10-05 by goxofy