landonr/homeThing #104: Error with

jorgeassuncao, 2024-03-07:
**Describe the bug**
installed ESPhome and after flashing i get this error on the log. PCF8574 is soldered in the right position.

**log**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c72
invalid header: 0xffffff7f
invalid header: 0xffffff7f
invalid header: 0xffffff7f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
[I][logger:359]: Log initialized
[I][homething.nowplaying.control:017]: set_media_player_group null 1
[W][homething.menu.display:010]: Boot setup
[I][app:029]: Running through setup()...
[I][i2c.arduino:183]: Performing I2C bus recovery
[   463][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[E][pcf8574:012]: PCF8574 not available under 0x20
[E][component:113]: Component pcf8574 was marked as failed.

landonr, 2024-03-10:
try turning on `  scan: true` and see if it shows up as another address

jorgeassuncao, 2024-03-10:
Ok, i've added `scan: true` and i'm getting this:

`[C][logger:447]: Logger:
[C][logger:448]:   Level: DEBUG
[C][logger:449]:   Log Baud Rate: 115200
[C][logger:451]:   Hardware UART: USB_CDC
[C][i2c.arduino:053]: I2C Bus:
[C][i2c.arduino:054]:   SDA Pin: GPIO18
[C][i2c.arduino:055]:   SCL Pin: GPIO17
[C][i2c.arduino:056]:   Frequency: 50000 Hz
[C][i2c.arduino:059]:   Recovery: bus successfully recovered
[I][i2c.arduino:069]: Results from i2c bus scan:
[I][i2c.arduino:075]: Found i2c device at address 0x38`

What do i have to change to make this work?

landonr, 2024-03-13:
you can edit substitutions for those lines like this
https://github.com/landonr/homeThing/blob/58bd93a40121b548a5b176bffd16370f56bc546f/tdisplay-s3.yaml#L78

jorgeassuncao, 2024-03-16:
Changed the substitutions code to:

`substitutions:
  friendly_name: "s3 homething"
  rotary_pin_a: "10"
  rotary_pin_b: "11"
  medium_font_size: "16"
  # sda_pin: "18"
  # scl_pin: "17"
  # screen_left_pin: "GPIO0"
  # screen_right_pin: "GPIO14"
  battery_pin: "4"
  # log_level: "DEBUG"
  launch_image_size: 100x100
  launch_image_file: "https://freight.cargo.site/w/450/i/6e34213b4c7e34285aa1729af3f7f0bd6c2d21412ca40407073d7ec8be40ed7c/E75A57A1-8ACD-4004-A603-647BF7AD04F7_1_105_c.jpeg"
  launch_image_type: "RGBA"

  # common/nowPlayingImage.yaml
  online_image_id: "media_player_image"
  online_image_initial_url: "http://172.16.1.23:8080/insecure/rs:fill:170:170/"
  online_image_text_sensor_id: "entity_picture"
  online_image_text_sensor_entity_id: "sensor.beam_picture_proxy"
  online_image_display_id: "my_display"
  
  # pcf8574
  address: "0x20"`
  
  But now when compiling i get this error:
  
  `INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/homething.yaml...
INFO Updating https://github.com/landonr/homeThing@main
INFO Updating https://github.com/landonr/lilygo-tdisplays3-esphome.git@v0.1
INFO Updating https://github.com/landonr/homeThing.git@None
INFO Updating https://github.com/landonr/esphome-components@main
INFO Detected timezone 'Europe/Lisbon'
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing lilygos3 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- pngle @ 1.0.0
|-- TFT_eSPI @ 2.5.43
Compiling .pioenvs/lilygos3/src/esphome/components/font/font.cpp.o
Compiling .pioenvs/lilygos3/src/esphome/components/homeThing/homeThingMenuBase.cpp.o
In file included from src/esphome/components/font/font.cpp:1:
src/esphome/components/font/font.h:53:8: error: 'void esphome::font::Font::print(int, int, esphome::display::Display*, esphome::Color, const char*, esphome::Color)' marked 'override', but does not override
   void print(int x_start, int y_start, display::Display *display, Color color, const char *text,
        ^~~~~
*** [.pioenvs/lilygos3/src/esphome/components/font/font.cpp.o] Error 1
In file included from src/esphome/components/homeThingDisplayState/homeThingDisplayState.h:5,
                 from src/esphome/components/homeThing/homeThingMenuHeader.h:6,
                 from src/esphome/components/homeThing/homeThingMenuBoot.h:9,
                 from src/esphome/components/homeThing/homeThingMenuBase.h:10,
                 from src/esphome/components/homeThing/homeThingMenuBase.cpp:1:
src/esphome/components/font/font.h:53:8: error: 'void esphome::font::Font::print(int, int, esphome::display::Display*, esphome::Color, const char*, esphome::Color)' marked 'override', but does not override
   void print(int x_start, int y_start, display::Display *display, Color color, const char *text,
        ^~~~~
*** [.pioenvs/lilygos3/src/esphome/components/homeThing/homeThingMenuBase.cpp.o] Error 1
========================== [FAILED] Took 4.65 seconds ==========================`

Do you have any idea?

landonr, 2024-03-24:
You can remove the `font` component because it's merged into ESPHome now. I updated the example config

jorgeassuncao, 2024-03-25:
Lots of errors when compiling...

```
INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/homething.yaml...
INFO Updating https://github.com/landonr/homeThing@main
INFO Updating https://github.com/landonr/lilygo-tdisplays3-esphome.git@v0.1
INFO Updating https://github.com/landonr/homeThing.git@None
INFO Updating https://github.com/landonr/esphome-components@main
INFO Detected timezone 'Europe/Lisbon'
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO3 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

font: [source /data/packages/c523de9d/common/fonts.yaml:7]
  - 
    Unknown value 'web', valid options are 'local', 'gfonts', 'local_bitmap'.
    file: 
      url: https://github.com/landonr/homeThing/raw/main/fonts/iosevka-heavy.ttf
      type: web
    id: large_heavy_font
    size: 24
    glyphs: |-
      ^…*/|\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓčëËüÜïÏöÖøØṣṢñãõÑÃÕ'’
  - 
    Unknown value 'web', valid options are 'local', 'gfonts', 'local_bitmap'.
    file: 
      url: https://github.com/landonr/homeThing/raw/main/fonts/iosevka.ttf
      type: web
    id: large_font
    size: 24
    glyphs: |-
      ^…*/|\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓčëËüÜïÏöÖøØṣṢñãõÑÃÕ'’
  - 
    Unknown value 'web', valid options are 'local', 'gfonts', 'local_bitmap'.
    file: 
      url: https://github.com/landonr/homeThing/raw/main/fonts/iosevka.ttf
      type: web
    id: medium_font
    size: 16
    glyphs: |-
      ^…*/|\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓčëËüÜïÏöÖøØṣṢñãõÑÃÕ'’
  - 
    Unknown value 'web', valid options are 'local', 'gfonts', 'local_bitmap'.
    file: 
      url: https://github.com/landonr/homeThing/raw/main/fonts/iosevka.ttf
      type: web
    id: small_font
    size: 14
    glyphs: |-
      ^…*/|\$#<>!?"%()[]+=,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz&êÊîÎâÂôÔéÉíÍáÁóÓčëËüÜïÏöÖøØṣṢñãõÑÃÕ'’
  - 
    Could not find file '/config/esphome/https://github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf'. Please make sure it exists (full path: /config/esphome/https:/github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf).
    file: |-
      https://github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf
    id: material_font_large
    size: 19
    glyphs: 
      - 󰐊
      - 󰓛
      - 󰏤
      - 󰽥
      - 󰒝
      - 󰒞
      - 󰕾
      - 󰕿
      - 󰚺
      - 󰝆
      - 󰗃
      - 󰓇
      - 󰔂
      - 󰋜
      - 󰌵
      - 󰌶
      - 󰑗
      - 󰑘
      - 󰑖
  - 
    Could not find file '/config/esphome/https://github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf'. Please make sure it exists (full path: /config/esphome/https:/github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf).
    file: |-
      https://github.com/landonr/homeThing/raw/main/fonts/materialdesignicons-webfont.ttf
    id: material_font_small
    size: 18
    glyphs: 
      - 󰐊
      - 󰓛
      - 󰏤
      - 󰽥
      - 󰒝
      - 󰒞
      - 󰕾
      - 󰕿
      - 󰚺
      - 󰝆
      - 󰗃
      - 󰓇
      - 󰔂
      - 󰋜
      - 󰌵
      - 󰌶
      - 󰑗
      - 󰑘
      - 󰑖
```

landonr, 2024-03-26:
`INFO ESPHome 2024.2.2` update to ESPHome 2024.3.0

closed 2024-05-10 by landonr

ssttaarr33, 2024-05-10:
> **Describe the bug** installed ESPhome and after flashing i get this error on the log. PCF8574 is soldered in the right position.
>
> **log**
>
> ```
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x15 [...]

Did you find out what was the issue? I'm facing the same one

jorgeassuncao, 2024-05-10:
Hi @ssttaarr33

The suggested substitutions code `address: "0x20"` on the `homeThing/tdisplay-s3.yaml` file did not work but the errors i had compiling disappeared after the update of ESPHome to v2024.3.0.

The only way i got it working was cloning the repo and change the file `/common/ipod/lilygo_tdisplay_ipod_binary_sensor_pcf.yaml` to use the correct address.

landonr, 2024-05-14:
> Hi @ssttaarr33
>
> The suggested substitutions code `address: "0x20"` on the `homeThing/tdisplay-s3.yaml` file did not work but the errors i had compiling disappeared after the update of ESPHome to [...]

thanks @jorgeassuncao I'll try to update the example to use variables instead of substitutions

jorgeassuncao, 2024-05-14:
Sorry for not giving feedback earlier...