landonr/homeThing #105: Error setting up the yaml

JoeGandy, 2024-03-22:
**Describe the bug**

Failed config

binary_sensor.gpio: [source /config/.esphome/packages/c523de9d/common/ipod/lilygo_tdisplay_ipod_binary_sensor_pcf.yaml:17]
  
  Pin 0 incorrectly sets allow_other_uses: true.
  platform: gpio
  pin: 
    number: 0
    allow_other_uses: True
    inverted: True

**To Reproduce**
Steps to reproduce the behavior:
1. Using default yaml: https://raw.githubusercontent.com/landonr/homeThing/main/tdisplay-s3.yaml

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

JoeGandy, 2024-03-22:
Potentially caused by https://github.com/landonr/homeThing/commit/416c33fd72800b0393d2b0d5d31eb39fc5a8230a

?

landonr, 2024-03-24:
@JoeGandy I updated the yaml.

if you want to use the config locally use can do this. The file has to be available locally so you can copy it into your esphome folder
```yaml
packages: # maybe modify: include default ipod controls
  -+ ipod_control_binary_sensor_pcf: !include ../homeThing/common/ipod/lilygo_tdisplay_ipod_binary_sensor_pcf.yaml
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [
      common/device_base.yaml, # defines api, ota, free memory and uptime sensor
      common/ipod/lilygo_tdisplay_ipod_battery.yaml, # used for battery percent
      # comment this out! common/ipod/lilygo_tdisplay_ipod_binary_sensor_pcf.yaml, # used for button controls
```

closed 2024-03-24 by landonr