landonr/homeThing #116: notifications

landonr, 2024-04-25:
**Is your feature request related to a problem? Please describe.**
notifications exist but theyre hard to call

**Describe the solution you'd like**
add service calls

**Additional context**
```yaml
voice_assistant:
  microphone: external_mic
  on_start:
    then:
      lambda: |-
        id(homeThingMenu)->clearNotifications();
        id(homeThingMenu)->addNotification("Voice Assistant", "Listening", "", true);
```

this should be a service call

landonr, 2024-04-28:
#118

closed 2024-04-28 by landonr

ManuRiver, 2024-07-26:
This sparked the following idea and wanted to share back with you guys.

Add the following self published service to esphome config file. 

services:   ##provisions a self published service back to homeassistant 
    - service: notify_homething
      variables:
        title: string
        message: string
        # type_str: string
      then:
        - lambda: |-
            id(homeThingMenu)->addNotification(title, message, "", true);

configuration.yaml on the hass side, profit ;-) 

notify:
  - platform: group
    name: "Notify Homething"
    services:
      - service: esphome ##propagates the notify.notify events back to the variables and calls back the service.
        data:
          target: studiotdisplay_notify_homething