landonr/homeThing #122: Error on compilation for StickCplus: component not found

Aaroneisele55, 2024-05-14:
**Describe the bug**
The compilation fails with the error "Component not found: homeThing."

**To Reproduce**
Steps to reproduce the behavior:
1. Use the M5StickCplus example code
2. try to compile
3. get the above error

**Expected behavior**
It should just work

**Logs**

[logs_m5stickcplus-homething_compile.txt](https://github.com/landonr/homeThing/files/15312209/logs_m5stickcplus-homething_compile.txt)

**Desktop (please complete the following information):**
 - OS: Windows
 - Browser Chrome
 - Version 11

**Additional context**
Add any other context about the problem here.

landonr, 2024-05-24:
can you share your full config? In the M5StickCPlus example homeThing is included here https://github.com/landonr/homeThing/blob/b976359ad3ab50d78f41c66a298985b5b1e51c60/m5stack-stickc.yaml#L26

we can see the project is building properly here https://github.com/landonr/homeThing/actions/runs/9216531658/job/25357000146

Aaroneisele55, 2024-05-24:
This is my config:
```
esphome:
  name: m5stickcexample
  friendly_name: m5stickcexample

esp32:
  board: m5stick-c
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

substitutions:
  friendly_name: "homeThingM5StickPlus"

external_components:
  - source:
      type: git
      url: https://gitlab.com/geiseri/esphome_extras.git
    refresh: 0s
    components: [axp192]
  - source:
      type: git
      url: https://github.com/landonr/homeThing
      ref: main
    refresh: 0s
    components: [homeThing]
  - source:
      type: git
      url: https://github.com/landonr/esphome-components
      ref: main
    refresh: 0s
    components: [
      homeassistant_component,
      homeassistant_media_player,
      media_player_source,
      media_player_source_sonos,
      media_player_source_spotify,
      media_player_source_custom
    ]

packages:
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [
      common/device_base.yaml, # defines api, ota, free memory and uptime sensor
      common/fonts.yaml, # default font
      common/icon_fonts.yaml # material icons
    ]
    refresh: 0s

i2c:
   - id: bus_a
     sda: GPIO21
     scl: GPIO22
     scan: true
   - id: bus_b
     sda: GPIO0
     scl: GPIO26
     scan: true

axp192:
  id: axp
  i2c_id: bus_a
  address: 0x34
  update_interval: 60s
  charge_voltage: 4150mV
  dcdc1_voltage: 3300mv
  dcdc3_voltage: 1.5V

sensor:
    # AXP192 power management - must be present to initialize TFT power on
  - platform: axp192
    axp192_id: axp
    id: battery_percent
    type: battery_power
    name: "${friendly_name} Battery Percent"

# internal LED
light:
  - platform: monochromatic
    output:  builtin_led
    name: Led
    id: led1
    internal: True
  - platform: monochromatic
    output: axp_ldo3
    name: "${friendly_name} Backlight"
    id: backlight
    restore_mode: ALWAYS_ON 

output:
  - platform: ledc
    pin: 10
    inverted: true
    id: builtin_led
  - platform: axp192
    axp192_id: axp
    output: ldo3
    id: axp_ldo3  

binary_sensor:
  - platform: axp192
    axp192_id: axp
    type: charge_indicate
    id: axp_charger
    name: "Charger"
  - platform: gpio
    pin:
      number: GPIO37
      inverted: true
    name: Button A
    on_click:
      min_length: 5ms
      max_length: 350ms
      then:
        lambda: |-
            id(homeThingMenu)->rotaryScrollClockwise(0);
            auto call = id(led1).turn_on();
            call.set_transition_length(0);
            call.perform();
    on_multi_click:
    - timing:
        - ON for at least 0.5s
      then:
      - lambda: |-
          id(homeThingMenu)->buttonPressUp();
    on_release:
      then:
        - light.turn_off: led1
  - platform: gpio
    pin:
      number: GPIO39
      inverted: true
    name: Button B
    on_press:
      then:
      - lambda: |-
          id(homeThingMenu)->buttonPressSelect();
          auto call = id(led1).turn_on();
          call.set_transition_length(0);
          call.perform();
    on_release:
      then:
        - light.turn_off: led1

deep_sleep:
  id: deep_sleep_
  wakeup_pin: 37
  wakeup_pin_mode: INVERT_WAKEUP

switch:
  - platform: axp192
    axp192_id: axp
    output: ldo3
    name: LDO3
    internal: true
  - platform: template
    name: "${friendly_name} Sleep Toggle"
    id: sleep_toggle
    optimistic: true
    on_turn_on:
      then:
        - light.turn_off:
            id: backlight
            transition_length: 0s
        - delay: 0.5s
        - deep_sleep.enter:
            id: deep_sleep_

spi:
  clk_pin: GPIO13
  mosi_pin: GPIO15

display:
  - platform: st7789v
    model: TTGO_TDisplay_135x240
    id: my_display
    cs_pin: GPIO5
    dc_pin: GPIO23
    reset_pin: GPIO18
    rotation: 0
    update_interval: 3600s
    lambda: |-
      id(homeThingMenu)->draw_menu_screen();
      return;

media_player_source_sonos:
  id: sonos

media_player_source_spotify:
  id: spotty

media_player_source_custom:
  - id: fav_playlists
    name: playlists
    sources:
      - name: Radio 7
        media_id: "aac://https://radio7.streamabc.net/87-radio7dab-aac-128-7982739?sABC=6602p43q%230%23q85q2494s3n583o622211rp6r70q6586%23gharva&aw_0_1st.playerid=tunein&amsparams=playerid:tunein;skey:1711457341"
        media_type: music
      - name: DasDing
        media_id: "aac://https://d121.rndfnk.com/ard/swr/dasding/live/aac/96/stream.aac?aggregator=tunein&cid=01FBVQWZT2B1KGPFJ7TDHQ1Y2B&sid=2eE69I1YfGMxvI15XlYS3F8eqJ2&token=MXV_qlHh0WjjvT3IXWvWcBYd-LzoJlCDkKklpaM2KIk&tvf=yjSm_O5kwBdkMTIxLnJuZGZuay5jb20"
        media_type: music

media_player:
  - platform: homeassistant_media_player
    name: Beam
    entity_id: "media_player.sonos_aaron_2"
    id: media_player_sonos_aaron
    type: speaker
    sources:
      - id: sonos
        type: sonos
      - id: spotty
        type: spotify
      - id: fav_playlists
        type: custom
homeassistant_media_player:
  id: media_group_component
  media_players:
    - id: media_player_sonos_aaron
      type: speaker

homeThing:
  id: homeThingMenu
  settings:
    mode: 2_button
  sleep_switch: sleep_toggle
  backlight: backlight
  display: my_display
  on_redraw:
    then:
      component.update: my_display
  display_state:
    font_small: small_font
    font_medium: medium_font
    font_large: large_font
    font_large_heavy: large_heavy_font
    font_material_large: material_font_large
    font_material_small: material_font_small
    font_logo: home_thing_logo
    draw_now_playing_bottom_menu: True
```

landonr, 2024-05-25:
@Aaroneisele55 thanks for sharing, the error its giving isn't that specific.
I moved DisplayState out of the homeThing config so it can be shared with other apps
the following should work
```yaml
esphome:
  name: m5stickcexample
  friendly_name: m5stickcexample

esp32:
  board: m5stick-c
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

substitutions:
  friendly_name: "homeThingM5StickPlus"

external_components:
  - source:
      type: git
      url: https://gitlab.com/geiseri/esphome_extras.git
    refresh: 0s
    components: [axp192]
  - source:
      type: git
      url: https://github.com/landonr/homeThing
      ref: main
      # type: local
      # path: ../homeThing/components
    refresh: 0s
    components: [homeThing, homeThingDisplayState, homeThingApp, homeThingAppNowPlaying, homeThingAppBreakout ]
  - source:
      type: git
      url: https://github.com/landonr/esphome-components
      ref: main
    refresh: 0s
    components: [
      homeassistant_component,
      homeassistant_media_player,
      media_player_source,
      media_player_source_sonos,
      media_player_source_spotify,
      media_player_source_custom
    ]

packages:
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [
      common/device_base.yaml, # defines api, ota, free memory and uptime sensor
      common/fonts.yaml, # default font
      common/icon_fonts.yaml # material icons
    ]
    refresh: 0s

i2c:
   - id: bus_a
     sda: GPIO21
[... 171 lines cut ...]

homeThingDisplayState:
  id: display_state_id
  font_small: small_font
  font_medium: medium_font
  font_large: large_font
  font_large_heavy: large_heavy_font
  font_material_large: material_font_large
  font_material_small: material_font_small
  draw_battery_level: True

homeThing:
  id: homeThingMenu
  display_state: display_state_id
  settings:
    mode: 2_button
  sleep_switch: sleep_toggle
  backlight: backlight
  display: my_display
  on_redraw:
    then:
      component.update: my_display
```

sorry for the late reply

Aaroneisele55, 2024-05-25:
Ok, so I have uploaded your config to the StickCplus and all I get on the display is a blue line...
[image]

I attached my logs to help resolve this...

[logs_m5stickcplus-homething_logs.txt](https://github.com/landonr/homeThing/files/15441765/logs_m5stickcplus-homething_logs.txt)

Best regards
Aaron Eisele

landonr, 2024-05-25:
@Aaroneisele55 thanks for trying that out! 
- I added the now playing app, and a few entities as an example to your config
- i found a bug with the now playing app not refreshing if the `finished_loading` sensor wasnt found

I tested and this config builds and shows the state of my media player named `beam`
```yaml
esphome:
  name: m5stickcexample
  friendly_name: m5stickcexample

esp32:
  board: m5stick-c
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

substitutions:
  friendly_name: "homeThingM5StickPlus"

external_components:
  - source:
      type: git
      url: https://gitlab.com/geiseri/esphome_extras.git
    refresh: 0s
    components: [axp192]
  - source:
      type: git
      url: https://github.com/landonr/homeThing
      ref: main
      # type: local
      # path: ../homeThing/components
    refresh: 0s
    components: [homeThing, homeThingDisplayState, homeThingApp, homeThingAppNowPlaying, homeThingAppBreakout ]
  - source:
      type: git
      url: https://github.com/landonr/esphome-components
      ref: main
    refresh: 0s
    components: [
      homeassistant_component,
      homeassistant_media_player,
      media_player_source,
      media_player_source_sonos,
      media_player_source_spotify,
      media_player_source_custom
    ]

packages:
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [
      common/device_base.yaml, # defines api, ota, free memory and uptime sensor
      common/fonts.yaml, # default font
      common/icon_fonts.yaml # material icons
    ]
    refresh: 0s

i2c:
[... 189 lines cut ...]
  font_material_large: material_font_large
  font_material_small: material_font_small
  draw_battery_level: True

homeThing:
  id: homeThingMenu
  display_state: display_state_id
  apps:
    - now_playing
  settings:
    mode: 2_button
  sleep_switch: sleep_toggle
  backlight: backlight
  display: my_display
  on_redraw:
    then:
      component.update: my_display
  home_screen: # added home screen with sleep toggle and backlight control as example
    name: stickcplus
    entities:
      - id: sleep_toggle
        type: switch
      - id: backlight
        type: light
```

Aaroneisele55, 2024-05-27:
OK, so it boots correctly now, but how can I go back from a page?
I tried button A, B, and long press on each of them, but I didn't succeed...

Thanks for your help
Aaron

Aaroneisele55, 2024-05-27:
Thanks for your help, now it boots!

How can I make it control a light that is connected to HA (entity_id: light.wled)?

Best regards
Aaron

landonr, 2024-05-28:
@Aaroneisele55 you may need to enable API control, if you go to `http://homeAssistant:8123/config/integrations/integration/esphome`, click `Configure` on your device, then check the box

Aaroneisele55, 2024-05-28:
> @Aaroneisele55 you may need to enable API control, if you go to `http://homeAssistant:8123/config/integrations/integration/esphome`, click `Configure` on your device, then check the box for **Allow the device to make Home Assistant service calls.**

I have enabled it, but it still doesn't work...
Also, the device shuts itself down after ~30 seconds, even when the charger is plugged in

landonr, 2024-05-29:
@Aaroneisele55 it might be crashing? If you have any error logs it might help.

I haven't tested a wled light, but sonos and a other types of lights are tested on my side

The stickcplus doesn't have much ram. I use my StickCPlus to control one led strip plugged into it, and to change light scenes using service calls, othertwise it runs out of memory. Can you try a simpler config and see if the device will stay on

Aaroneisele55, 2024-05-29:
> @Aaroneisele55 it might be crashing? If you have any error logs it might help.
>
> I haven't tested a wled light, but sonos and a other types of lights are tested on my side
>
> The stickcplus doesn't [...]

What would that simpler config be?
I only really need control of 6 lights and one input_boolean, as well as status display of four binary sensors

closed 2024-07-31 by landonr