landonr/homeThing #132: Media player online image is not generated

iondulgheru, 2024-07-04:
Media player online image is not generated
I added in HA the template sensor to create the sensor media_player_image_proxy_url.
But this template should get the attribute entity_picture from a sensor called active_player.
No such sensor is defined in HA, I see that is defined in the HomeThing code, but is not exported to HA.
If I change the template to get this attribute from the the media_player I am using image generation works
I think the active_player sensor should also be exposed to HA.

landonr, 2024-07-05:
youre right it's missing from the example code! I added it in #133, it's like this

```
homeassistant_media_player: # media player menu
  id: media_group_component
  finished_loading:
    id: media_players_loaded_sensor
  active_player: 
    id: active_player
    name: "${friendly_name} Active Player"
  media_players:
    - id: media_player_beam
      type: speaker
    - id: media_player_tv
      type: roku
```

iondulgheru, 2024-07-05:
Also this "https://" concatenation breaks the link, which looks like "https://http://[...]", if it the HA template is defined according to the example
https://github.com/landonr/homeThing/blob/main/common/nowPlayingImage.yaml#L44

landonr, 2024-07-05:
I have to remove the `http` from the example url. Last time I checked, the online image code couldn't handle unencrypted urls (http). Is `http://` being added anywhere else?

iondulgheru, 2024-07-05:
In my case it works without any issues over http. Right now I have defined the full url (with http) in the HA template, and on esphome I removed the https concatenation. In my setup it doesn't work over https, I tried it.
These are the only changes I made, and it works.

landonr, 2024-07-05:
@iondulgheru I http is fixed! Http and https are both working for external images now. I updated #133 with updated instructions and i'll try to add a readme to https://github.com/landonr/esphome-components/tree/main/components/homeassistant_media_player
@guillempages

closed 2024-07-05 by landonr

iondulgheru, 2024-07-05:
Thank you @landonr 🙏