landonr/homeThing #136: m5 Stick Config Failing: "Unable to find action with the name 'homething_menu.scroll_counter_clockwise'."

barndawgie, 2024-07-12:
I'm trying to install homeThing to a brand new m5Stack FIRE device. When I try to validate the configuration in ESPHome Dashboard, I'm getting a failure: "Unable to find action with the name 'homething_menu.scroll_counter_clockwise'."

Here is my config, heavily based on your wiki and samples:

```yaml
---
substitutions:
  name: "home-thing"
  friendly_name: "homeThing"
  device_description: esp32 homeThing Remote Control
  board: m5stack-fire
  variant: esp32

packages:
  remote_package:
    url: https://github.com/landonr/homeThing
    ref: main
    files: [ # See https://github.com/landonr/homeThing/blob/main/GettingStarted.md
      common/device_base.yaml, # defines api, ota, free memory and uptime sensor
      common/m5stack-fire/backlight.yaml, # used for toggling backlight
      common/m5stack-fire/binary_sensor.yaml, # used for button controls
      common/m5stack-fire/binary_sensor_rotary.yaml, # used for rotary controls
      common/m5stack-fire/ip5306.yaml, # used for battery percent
      common/m5stack-fire/sidelight.yaml, # used to control LED strip
      common/fonts.yaml, # default font
      common/icon_fonts.yaml, # material icons
      common/images.yaml, # boot screen image
    ]
  base_config: !include config_deps/base_config.yaml

external_components:
  - source: # homeThing, display setup, and now playing app
      type: git
      url: https://github.com/landonr/homeThing
      ref: main
    refresh: 0s
    components: [
      homeThing, 
      homeThingDisplayState,
      homeThingApp, # only required if using apps
      homeThingAppNowPlaying # only required for Now Playing app
    ]
  - source:
      type: git
      url: https://github.com/landonr/esphome-components
      ref: main
    refresh: 0s
    components: [
      homeassistant_component,
      homeassistant_media_player,
      media_player_source,
      media_player_source_sonos,
      media_player_source_spotify,
      media_player_source_custom
    ]
  - source:
      type: git
      url: https://github.com/ssieb/custom_components
    components: [ ip5306 ]

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  comment: ${device_description}

esp32:
  board: ${board}
  framework:
    type: esp-idf
  variant: ${variant}
```

Here is what I get from validate using ESPHome 2024.6.6 Dashboard:
```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/home-thing.yaml...
INFO Updating https://github.com/landonr/homeThing@main
INFO Updating https://github.com/landonr/esphome-components@main
INFO Detected timezone 'America/Los_Angeles'
Failed config

binary_sensor.gpio: [source /config/.esphome/packages/c523de9d/common/m5stack-fire/binary_sensor.yaml:2]
  platform: gpio
  id: M5_BtnA
  pin: 
    number: 39
    inverted: True
  on_click: 
    then: 
      
      Unable to find action with the name 'homething_menu.scroll_counter_clockwise'.
      homething_menu.scroll_counter_clockwise: homeThingMenu
```

I don't _think_ this particular error can be caused by something on my side, but maybe I am missing something...

barndawgie, 2024-07-18:
Thanks for the reply!

>you shouldn't need both of the binary sensor configs. If you don't have the external encoder, just use the binary_sensor otherwise use the rotary one

Makes sense. I removed `binary_sensor_rotary.yaml` but still get the error.

>i noticed base_config: !include config_deps/base_config.yaml can you share this? It might have includes which overwrite includes in the config you shared. For example if you define images in the base config, and then in the package, it's overwritten in the base

This is my own private base config - it just configures WiFi and sets the API Encryption key. I removed it and moved these into the config for debugging simplicity.

> if you remove the binary_sensor.yaml include, are you able to compile and boot into the menu?

No. if I remove `binary_sensor.yaml` I get errors saying that "Component font requires component display." If I remove every package accept `device_base.yaml` I still get errors. Seems like these may also be related to the lack of an m5stack dispaly in the configured files?

```
Failed config

api: [source /config/.esphome/packages/c523de9d/common/device_base.yaml:19]
  id: homeassistant_api_id
  services: 
    - service: goto_screen
      variables: 
        screen_name: string
      then: 
        - 
          Couldn't find ID 'homeThingMenu'. Please check you have defined an ID with that name in your configuration.
          lambda: !lambda |-
            id(homeThingMenu)->goToScreenFromString(screen_name);
  encryption: 
    key: Rjtway9CYmFqUGNQMVtAOERKI2h7Yz1ZdVgtMDh5KGY=
  port: 6053
  password: ''
time.homeassistant: [source /config/.esphome/packages/c523de9d/common/device_base.yaml:30]
  platform: homeassistant
  id: esptime
  on_time_sync: 
    - then: 
        - component.update: 
            
            Couldn't find ID 'my_display'. Please check you have defined an ID with that name in your configuration.
            id: my_display
  timezone: PST8PDT,M3.2.0,M11.1.0
  update_interval: 15min
```

barndawgie, 2024-07-18:
Seems like I missed a lot of components, really. I was trying to base this off of this wiki (https://github.com/landonr/homeThing/wiki/m5stack-fire) but it contains a mix of what seem like required components and references to your own setup.

Let me spend some time and try to figure out an "empty" m5stack config and maybe I can submit in a PR or something.

landonr, 2024-07-19:
That wiki page is pretty out of date, i'm going to remote it!
I have this example file https://github.com/landonr/homeThing/blob/main/m5stack-fire.yaml which should build. update the entities in there
There's this basic config you can compare it against https://github.com/landonr/homeThing/blob/main/basic.yaml

this is my personal M5Stack Fire config that I use to control my cat's auto feeder https://github.com/landonr/esphome-devices/blob/main/fireremote.yaml